gazebosim/gazebo_test_cases #1513: gz-sensors: Segmentation Camera in Gazebo

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/sensors/9/segmentationcamera_igngazebo.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/segmentation_camera.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2025-08-25:
test 1

bperseghetti, 2025-08-25:
*pass*

bperseghetti, 2025-08-25:
**passed**

bperseghetti, 2025-08-25:
> **failed**

That's so weird that mine actually passed as you can see 

> **passed**

gbiggs, 2025-08-26:
**passed**

gbiggs, 2025-08-26:
[status: passed]

gbiggs, 2025-08-26:
pass pass passity pass

gbiggs, 2025-08-26:
[image]

gbiggs, 2025-08-26:
[image]

mjcarroll, 2025-08-26:
[status: passed]