gazebosim/gazebo_test_cases #1514: gz-sensors: Segmentation Camera in Gazebo

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/sensors/9/segmentationcamera_igngazebo.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/segmentation_camera.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

csimon-be, 2024-08-30:
Interested in this one

akky20, 2024-09-02:
### Simulation works properly we are getting list of topics for this   
[image]

### Topics 
[image]

ahcorde, 2024-09-03:
Thank you for testing!

closed 2024-09-03 by ahcorde