gazebosim/gazebo_test_cases #1515: gz-sensors: Segmentation Camera in Gazebo

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/sensors/9/segmentationcamera_igngazebo.html)
* [Source](https://github.com/gazebosim/gz-sensors/tree/gz-sensors9/tutorials/segmentation_camera.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-16:
Little fixes in https://github.com/gazebosim/gz-sensors/pull/471 .

Other from that, the tutorial works. I've also tested the Python script and the C++ snippet. This issue can be closed.

ahcorde, 2024-09-16:
PR is merged, thank you for testing!

closed 2024-09-16 by ahcorde