gazebosim/gazebo_test_cases #1520: gz-tools: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/tools)
* [Source](https://github.com/gazebosim/gz-tools/blob/gz-tools2/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alexandrosnic, 2024-09-03:
Interested in this one

alexandrosnic, 2024-09-03:
The installation tutorial of both the [Live](https://gazebosim.org/libs/tools/) and [Source](https://github.com/gazebosim/gz-tools/blob/gz-tools2/README.md) page leads to "**The page is not found.**".

Other than that, looks good.

alexandrosnic, 2024-09-05:
@azeey  it seems that the gazebosim's installation tutorial page for gz-tools was not published correctly. Can I fix this through the repository?

azeey, 2024-09-05:
Thanks for testing. The links are now working.

closed 2024-09-05 by azeey