gazebosim/gazebo_test_cases #1523: gz-tools: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/tools/2/install.html)
* [Source](https://github.com/gazebosim/gz-tools/tree/gz-tools2/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-29:
## Binary Installation
To download and store GPG keys (apt-key format is deprecated and hence trusted.gpg.d should be used instead):
```
wget -qO - http://packages.osrfoundation.org/gazebo.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/gazebo-archive-keyring.gpg > /dev/null
```

Check available version of libgz-tools:
```
apt-cache search libgz-tools
```

caguero, 2024-09-03:
Thanks for the help! could you create a PR with your suggestions?

ahcorde, 2024-09-05:
@trushant05 friendly ping

trushant05, 2024-09-05:
Hello @caguero  and @ahcorde , just submitted [PR 144](https://github.com/gazebosim/gz-tools/pull/144). Do let me know if something needs to be changed. Thanks.