gazebosim/gazebo_test_cases #1524: gz-tools: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/tools/2/install.html)
* [Source](https://github.com/gazebosim/gz-tools/tree/gz-tools2/tutorials/install.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-11:
I'll test this over the weekend on a real Win 10 laptop.

Combuster54, 2024-09-12:
## Windows 10

**Binary Installation**

```sh
(gz-ws) C:\Users\User\gazebo-ws\test>conda search libgz-tools* --channel conda-forge
Loading channels: done
# Name                       Version           Build  Channel
libgz-tools2                   2.0.0      h63175ca_0  conda-forge
libgz-tools2                   2.0.0      h63175ca_1  conda-forge
libgz-tools2                   2.0.0      hc79a705_2  conda-forge
libgz-tools2                   2.0.0      hdf75bb1_2  conda-forge
libgz-tools2                   2.0.0      hdf75bb1_3  conda-forge
libgz-tools2                   2.0.0      hdf75bb1_4  conda-forge
libgz-tools2                   2.0.0      hf5993cb_3  conda-forge
libgz-tools2                   2.0.0      hf5993cb_4  conda-forge
libgz-tools2                   2.0.1      h9fbcf1e_0  conda-forge

(gz-ws) C:\Users\User\gazebo-ws\test>conda install libgz-tools2 --channel conda-forge
Retrieving notices: ...working... done
Channels:
 - conda-forge
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: C:\Users\User\anaconda3\envs\gz-ws

  added / updated specs:
    - libgz-tools2

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    libgz-cmake3-3.5.3         |       he0c23c2_0         200 KB  conda-forge
    libgz-tools2-2.0.1         |       h9fbcf1e_0          44 KB  conda-forge
    ------------------------------------------------------------
                                           Total:         244 KB

The following NEW packages will be INSTALLED:

  libgz-cmake3       conda-forge/win-64::libgz-cmake3-3.5.3-he0c23c2_0
  libgz-tools2       conda-forge/win-64::libgz-tools2-2.0.1-h9fbcf1e_0

Proceed ([y]/n)? y

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
```

**Source Instalation**

```sh

(gz-ws) C:\Users\User\gazebo-ws\test>conda search libgz-tools* --channel conda-forge --info
Loading channels: done
libgz-tools2 2.0.0 h63175ca_0
-----------------------------
file name   : libgz-tools2-2.0.0-h63175ca_0.tar.bz2
name        : libgz-tools2
version     : 2.0.0
build       : h63175ca_0
build number: 0
[... 244 lines cut ...]
-- Found Backward: C:/Users/User/gazebo-ws/test/gz-tools/src/Backward
-- The program [cppcheck] was not found! Skipping codecheck setup
-- Build configuration successful
-- Build type: RelWithDebInfo
-- Install prefix: C:/Program Files (x86)/gz-tools2
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Configuring done (5.8s)
-- Generating done (0.1s)
-- Build files have been written to: C:/Users/User/gazebo-ws/test/gz-tools/build

(gz-ws) C:\Users\User\gazebo-ws\test\gz-tools\build>cmake --build . --config Release
Versión de MSBuild 17.11.2+c078802d4 para .NET Framework

  1>Checking Build System
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-tools/src/CMakeLists.txt
  backward.cc
  backward.obj : MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link
  command line to improve linker performance
  Generating code
  Finished generating code
  backward.vcxproj -> C:\Users\User\gazebo-ws\test\gz-tools\build\bin\Release\gz-tools2-backward.dll
  Building Custom Rule C:/Users/User/gazebo-ws/test/gz-tools/CMakeLists.txt

(gz-ws) C:\Users\User\gazebo-ws\test\gz-tools\build>
```

closed 2024-09-13 by azeey