gazebosim/gazebo_test_cases #1526: gz-transport: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/transport)
* [Source](https://github.com/gazebosim/gz-transport/blob/gz-transport14/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-09-02:
Interested

Mechazo11, 2024-09-04:
@azeey-test I may not get enough time to test this one. please unassign me from this one.

Sanjay1911, 2024-09-04:
I can check the read me and update @azeey

Sanjay1911, 2024-09-04:
- [ ] Binary Install:

Followed the steps for binary install and encountered the below issues. 
**sudo apt-get install libgz-transport9-dev**
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package libgz-transport9-dev
  
  **sudo apt-get install libgz-transport8-dev**
  Reading package lists... Done
  Building dependency tree... Done
  Reading state information... Done
  E: Unable to locate package libgz-transport8-dev

- [x]  Source Install

Encountered  **error** when installing prerequisites. Removed "python" and worked as expected.
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Package python is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> However the following packages replace it:
>   2to3 python-is-python3

- [x] Tests
[image]

j-rivero, 2024-09-05:
Looks good to me.

closed 2024-09-05 by j-rivero