gazebosim/gazebo_test_cases #1540: gz-transport: Services

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/transport/14/services.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport14/tutorials/05_services.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

sreekanthr, 2024-09-11:
I would like to pick this for testing in Mac OS Monterey (12.7.6) [amd64]

sreekanthr, 2024-09-11:
No issues found.

### Build
[image]

### Test runs
[image]

closed 2024-09-12 by azeey