gazebosim/gazebo_test_cases #1543: gz-transport: Python Support

azeey-test, 2024-08-29:
## Configuration

* os: MacOS

## Links

* [Live](https://gazebosim.org/api/transport/14/python.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport14/tutorials/06_python_support.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

codingmartian, 2024-09-11:
I would like to pick this up

sreekanthr, 2024-09-11:
I would like to pick this for testing in Mac OS Monterey (12.7.6) [amd64]

sreekanthr, 2024-09-11:
### Publisher / Subscriber
[image]

###  Threading in Gazebo Transport
[image]
[image]

###  Advertise Options
[image]

###  Subscribe Options
[image]

###  Topic remapping
[image]

###  Service Requester & Responser
[image]

sreekanthr, 2024-09-11:
Creating a PR for the below:

### Advertise Options
- The code provided in the tutorial needs minor change

### Topic Remapping:
- Code excerpt needs minor change

closed 2024-09-11 by azeey