gazebosim/gazebo_test_cases #1545: gz-transport: Python Support

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/transport/14/python.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport14/tutorials/06_python_support.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-15:
I couldn't get Python support to work on Windows.

I had to install protobuf, as only libprotobuf is installed according to the win install tutorial. This decreases libprotobuf version to 4.* from 5.*, as protobuf is only available in version 4.* . I rebuilt the workspace after this downgrade.

Now I see:

```
 λ dir D:\programming\gz9-ws\gz-ws\install\lib\python\gz\transport14
 Directory of D:\programming\gz9-ws\gz-ws\install\lib\python\gz\transport14

15.09.2024  23:28    <DIR>          .
15.09.2024  23:28    <DIR>          ..
15.09.2024  23:02           255 488 _transport.cp312-win_amd64.pyd
15.09.2024  23:28             5 529 __init__.py
```

I set PYTHONPATH to 

    set PYTHONPATH=D:\programming\gz9-ws\gz-ws\install\lib\python;D:\programming\gz9-ws\conda\Lib\site-packages

When I run the example, I get the following error:

```
λ python publisher.py
Traceback (most recent call last):
  File "D:\programming\gz9-ws\gz-ws\src\gz-transport\python\examples\publisher.py", line 19, in <module>
    from gz.transport14 import Node
  File "D:\programming\gz9-ws\gz-ws\install\lib\python\gz\transport14\__init__.py", line 16, in <module>
    from ._transport import Node as _Node
ImportError: DLL load failed while importing _transport: module not found
```

I've inspected the pyd file with Depends and nothing seems to be missing.

I can't continue with this here. Any thoughts, @traversaro?

traversaro, 2024-09-15:
Just a curiosity, why do you need protobuf? Anyhow, I think the problem is that Python on Windows by default ignores the PATH env variable when loading extensions (see https://github.com/robotology/robotology-superbuild/issues/1268 and the linked issue for some references).

Possible solution are calling `set CONDA_DLL_SEARCH_MODIFICATION_ENABLE=1` before launching python or call os.add_dll_directory with the directory of the location where the .dll of gZ-transport are stored.

peci1, 2024-09-15:
> Just a curiosity, why do you need protobuf?

Because of the content of the generated .py files from gz-msgs, e.g.

    #file gz/msgs/stringmsg_pb2.py
    from google.protobuf import descriptor as _descriptor
    ...

> Possible solution are calling `set CONDA_DLL_SEARCH_MODIFICATION_ENABLE=1` before launching python

Did not help.

> call os.add_dll_directory with the directory of the location where the .dll of gZ-transport are stored.

Did help! Now I can run the tutorial.

```
import os
os.add_dll_directory("D:\\programming\\gz9-ws\\gz-ws\\install\\bin")
from ._transport import Node as _Node
```

peci1, 2024-09-15:
I've reported the issue: https://github.com/gazebosim/gz-transport/issues/538 .

peci1, 2024-09-15:
A little improvement in https://github.com/gazebosim/gz-transport/pull/539 .

Required install tutorial change: https://github.com/gazebosim/docs/pull/516 .

Requires https://github.com/gazebosim/gz-transport/issues/538 to be fixed.

After the referenced PRs/issues are done, this tutorial works as it should.

closed 2024-09-16 by ahcorde