gazebosim/gazebo_test_cases #1551: gz-transport: Relay

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/transport/14/relay.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport14/tutorials/08_relay.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

azeey, 2025-08-26:
Testing point system:
```
caguero@bb9:~/gz_transport_tutorial/docker$ ifconfig
docker0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        inet6 fe80::42:73ff:fe1c:351e  prefixlen 64  scopeid 0x20<link>
        ether 02:42:73:1c:35:1e  txqueuelen 0  (Ethernet)
        RX packets 943154  bytes 72074740 (72.0 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1646253  bytes 2714146135 (2.7 GB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Result: **passed**

bperseghetti, 2025-08-26:
This seems a different result than mine

**failed**