gazebosim/gazebo_test_cases #1554: gz-transport: Logging

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/transport/14/logging.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport14/tutorials/10_logging.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-15:
`gz log` is not installed on windows due to https://github.com/gazebosim/gz-transport/blob/8b184b3ca001b67c4854ad8d3dd40f6f47cd29c6/log/src/CMakeLists.txt#L62, so the last part of the tutorial doesn't work on Windows.

Otherwise, the tutorial works.

azeey, 2024-09-16:
I've created an issue https://github.com/gazebosim/gz-transport/issues/540

closed 2024-09-16 by azeey