gazebosim/gazebo_test_cases #1566: gz-transport: Topic Statistics

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/transport/14/topicstatistics.html)
* [Source](https://github.com/gazebosim/gz-transport/tree/gz-transport14/tutorials/23_topic_statistics.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

peci1, 2024-09-11:
I'll test this over the weekend on a real Win 10 laptop.

j-rivero, 2024-09-12:
Using a from source installation on Windows, the example is not trivial to build on Windows without the CMakeLists.txt. Need to patch the documentation to use the examples directory CMakeLists.txt.

j-rivero, 2024-09-12:
Functionality works.

j-rivero, 2024-09-12:
PR https://github.com/gazebosim/gz-transport/pull/533