gazebosim/gazebo_test_cases #1571: gz-utils: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/api/utils/3/install.html)
* [Source](https://github.com/gazebosim/gz-utils/tree/gz-utils3/tutorials/installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-29:
## Binary Installation

For setup keys section, since apt-key is deprecated following could be used:
```
wget -qO - http://packages.osrfoundation.org/gazebo.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/gazebo-archive-keyring.gpg > /dev/null
```

And to check available version of gz-utils, following could be used:
```
apt-cache search
```

## Source Installation
Everything works fine

## Documentation
Everything works fine

## Testing
Everything works fine but for static code check, currently there are not rules to make.

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde