gazebosim/gazebo_test_cases #1572: gz-utils: Installation

azeey-test, 2024-08-29:
## Configuration

* os: Windows

## Links

* [Live](https://gazebosim.org/api/utils/3/install.html)
* [Source](https://github.com/gazebosim/gz-utils/tree/gz-utils3/tutorials/installation.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

jc-sz, 2024-08-30:
Interested in trying this out. :)

jc-sz, 2024-08-31:
In this [link](https://gazebosim.org/api/utils/3/install.html), it says replace <#> with a number 1 or 2.

"libgz-utils2" works fine, and it is available in [conda-forge](https://prefix.dev/channels/conda-forge/packages/libgz-utils2).

However, I am seeing the following PackageNotFoundError with trying to install "libgz-utils" or "libgz-utils1":

---
(gz-ws) D:\gz_bash\gz-utils\build>conda install libgz-utils1 --channel conda-forge
Channels:
 - conda-forge
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - libgz-utils1

Current channels:

  - https://conda.anaconda.org/conda-forge
  - defaults

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
---

Are they available on Windows? Or am I missing anything here?

j-rivero, 2024-09-12:
> Are they available on Windows? Or am I missing anything here?

Only gz-utils2 is available on Windows. Actually, if we look at the URL of the documentation, Ionic is using gz-utils3, so we can hardcode the number. The problem is that we don't have pre-releases for Windows so we can not test this method. Releases will probably be in conda-forge in the next weeks.

Only the from-source method can be used.