gazebosim/gazebo_test_cases #1574: sdformat: README

azeey-test, 2024-08-29:
## Configuration

* os: Ubuntu Noble

## Links

* [Live](https://gazebosim.org/libs/sdformat)
* [Source](https://github.com/gazebosim/sdformat/blob/sdf15/README.md)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-29:
I am interested to test this

trushant05, 2024-08-29:
I have tested the binary installation on Ubuntu 24.04 Noble and I would like to suggest some changes which are as follow:

Step 1 : Adding the repository is correct:
```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" > /etc/apt/sources.list.d/gazebo-stable.list'
```

Step 2 : Download and store GPG keys (apt-key format is deprecated and hence trusted.gpg.d should be used instead):
```
wget -qO - http://packages.osrfoundation.org/gazebo.key | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/gazebo-archive-keyring.gpg > /dev/null
```

Step 3 : Check available version of sdformat-dev:
```
apt-cache search libsdformat
```

Step 4 : Install sdformat
```
sudo apt install libsdformat<#>-dev libsdformat<#>
```

trushant05, 2024-08-29:
@azeey would the above format for test suffice or anything specific is required?

azeey, 2024-08-30:
Thanks for testing. The format is great! For the suggested fix, it would be awesome if you could create a PR.

closed 2024-08-30 by azeey