gazebosim/gazebo_test_cases #1576: gz-sim: 3k_shapes.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/3k_shapes.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

Mechazo11, 2024-09-02:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full Desktop 
    ```
* **Summary**
    * Over 15GB RAM usage, no shapes generated. **Could not get to generate shapes.** The suggested command `erb -T 1 shapes_population.sdf.erb > shapes_population.sdf` shows the following error

    ```terminal
    az-ubuntu-2204@az-dev-ubuntu-2404:~$ erb -T 1 shapes_population.sdf.erb > shapes_population.sdf
    /usr/lib/ruby/gems/3.2.0/gems/erb-4.0.2/libexec/erb:120:in `set_encoding': No such file or directory @ rb_sysopen - shapes_population.sdf.erb (Errno::ENOENT)
        from /usr/lib/ruby/gems/3.2.0/gems/erb-4.0.2/libexec/erb:120:in `run'
        from /usr/lib/ruby/gems/3.2.0/gems/erb-4.0.2/libexec/erb:164:in `<top (required)>'
        from /usr/bin/erb:25:in `load'
        from /usr/bin/erb:25:in `<main>'
    ```

* **Issues**
    * Failed to generate shapes, UI got laggy and no asset generated on the UI window. Suggested command throws an error message.
    
* **Fixes**
    * N/A

* **Comments/Suggestions**
    * N/A

arjo129, 2024-09-03:
> * **Test enviornment**
>   ```
>   CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
>   Clean copy of Ubuntu 24.04 Humble
>   ROS 2 Jazzy Jalisco Full Desktop 
>   ```
> * **Summary**
>   
>   * Over 15GB RAM usage, no shapes generated. **Could not get to generate shapes.** The suggested command `erb -T 1 shapes_population.sdf.erb > shapes_population.sdf` shows the following error
>   
>   ```
>   az-ubuntu-2204@az-dev-ubuntu-2404:~$ erb -T 1 shapes_population.sdf.erb > shapes_population.sdf
>   /usr/lib/ruby/gems/3.2.0/gems/erb-4.0.2/libexec/erb:120:in `set_encoding': No such file or directory @ rb_sysopen - shapes_population.sdf.erb (Errno::ENOENT)
>       from /usr/lib/ruby/gems/3.2.0/gems/erb-4.0.2/libexec/erb:120:in `run'
>       from /usr/lib/ruby/gems/3.2.0/gems/erb-4.0.2/libexec/erb:164:in `<top (required)>'
>       from /usr/bin/erb:25:in `load'
>       from /usr/bin/erb:25:in `<main>'
>   ```
> * **Issues**
>   
>   * Failed to generate shapes, UI got laggy and no asset generated on the UI window. Suggested command throws an error message.
> * **Fixes**
>   
>   * N/A
> * **Comments/Suggestions**
>   
>   * N/A

It doesn't work because you are doing it from your home directory. You need to be in the xamples directory for this to work.

Mechazo11, 2024-09-03:
@arjo129 thank you for the feedback, I will try this tutorial again today (if possible) and report back.

Mechazo11, 2024-09-04:
@arjo129 can you share the steps you took to get the demo working. Mine is crashing. Here are the steps I took

```bash
cd ~/workspace/src/gz-sim/examples/worlds
erb -T 1 shapes_population.sdf.erb > shapes_population.sdf
gz sim shapes_population.sdf
```

A gazebo sim window opens up, RAM usages goes upto 14 GB but nothing else happens. Waited about 3-5 minutes before a message to Force Quit Gazebo sim window popped up.

[image]

arjo129, 2024-09-04:
Can you run with the `-v4` option.

Mechazo11, 2024-09-04:
@arjo129 can you give me the full command please. Sorry but kinda very new to Gazebo not familiar with all the command structure yet.

arjo129, 2024-09-04:
```
cd ~/workspace/src/gz-sim/examples/worlds
erb -T 1 shapes_population.sdf.erb > shapes_population.sdf
gz sim -v4 shapes_population.sdf
```

Mechazo11, 2024-09-04:
Thanks @arjo129 I will try your commands and report back tomorrow

arjo129, 2024-09-05:
So I am able to get this working but it needs a lot of RAM. The server starts up relatively quickly, but the gui takes a very long time to load. The SDF parser takes ~30 seconds to start while the shapes take 10 minutes to appear on the gui screen. Total memory usafe is about 10GB of ram and 20GB of swap, I can see why most people would have trouble running this.

arjo129, 2024-09-05:
I've opened https://github.com/gazebosim/sdformat/issues/1478. I'm not an expert on libsdformat, but the root cause does seem like this. In the mean time https://github.com/gazebosim/gz-sim/pull/2596 can be used to reduce load times.

utkuesen, 2024-09-06:
I tested it and it worked for me. However, rendering process is taken so much time.
I tried it on machine which is HP ZBook 15 G3(i7-6820HQ , 32 Gb RAM, NVidia Quadro M1000M, Ubuntu 24.04.1 LTS) without docker .
I started program from home directory with this command "**gz sim /usr/share/gz/gz-sim9/worlds/3k_shapes.sdf**".
It took at least 5 min to render the all scene.
When i tried with this command "**gz sim /usr/share/gz/gz-sim9/worlds/3k_shapes.sdf -v4**", I get this warning till scene is created from beginning "(2024-09-06 19:22:46.933) [warning] [GUI] Unable to deserialize sdf::Model".

First minutes of the screen:
[image]

The final result:
[image]

arjo129, 2024-09-06:
Thanks @utkuesen, yeah the rendering is taking very long. We also know why as can be seen in my previous comments. We just need to come to a consensus on whats the best way forward. The current issue is the gui is taking too long to deserialize the ecm.

Mechazo11, 2024-09-06:
Sorry, clicked "unassign me" by mistake.

jennuine, 2024-09-06:
> Sorry, clicked "unassign me" by mistake.

I've re-assigned you 😄

closed 2024-09-16 by azeey