gazebosim/gazebo_test_cases #1579: gz-sim: acoustic_comms_demo.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/acoustic_comms_demo.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-31:
Interested

khshmt, 2024-08-31:
## Ubuntu 22.04 (jammy) ##
[image]
[Screencast from 08-31-2024 08:00:27 PM.webm](https://github.com/user-attachments/assets/fe2b96d2-f10a-4794-afa8-824b46bab79a)

closed 2024-09-03 by arjo129