gazebosim/gazebo_test_cases #1587: gz-sim: apply_joint_force.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/apply_joint_force.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-30:
I would like to test the tutorial contained in the issue.

avanmalleghem, 2024-08-30:
Commands
`gz topic -t "/model/joint_force_example/joint/j1/cmd_force" -m gz.msgs.Double  -p "data: 0.001"`
`gz topic -t "/model/joint_force_example/joint/j1/cmd_force" -m gz.msgs.Double  -p "data: 0.0"`
Work as expected

There is no images so this issue can be closed.

closed 2024-08-30 by iche033