gazebosim/gazebo_test_cases #1591: gz-sim: auv_controls.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/auv_controls.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-31:
Interested

Combuster54, 2024-09-02:
## Ubuntu 24.04 Noble

As I mention in https://github.com/gazebosim/gazebo_test_cases/issues/1305

**Error**

```bash
combuster@combuster54:~$ gz sim auv_controls.sdf
(2024-09-02 16:11:32.239) [error] The use of added mass through this plugin is deprecated and willbe removed in Gazebo J* as this formulation has instabilities. We recommend using the SDF `<fluid_added_mass>` tag based method[http://sdformat.org/spec?ver=1.11&elem=link#inertial_fluid_added_mass]To get rid of this warning we recommend setting`<disable_added_mass> to true.
```.
**1. Move the rudder**

**2. Propel the vehicle**

gz topic -t /model/tethys/joint/propeller_joint/cmd_thrust
      -m gz.msgs.Double -p 'data: -31'

Combuster54, 2024-09-02:
**1. Move the rudder**
[image]

**2. Propel the vehicle**
[image]

arjo129, 2024-09-03:
> ## Ubuntu 24.04 Noble
> 
> As I mention in #1305
> 
> **Error**
> 
> ```shell
> combuster@combuster54:~$ gz sim auv_controls.sdf
> (2024-09-02 16:11:32.239) [error] The use of added mass through this plugin is deprecated and willbe removed in Gazebo J* as this formulation has instabilities. We recommend using the SDF `<fluid_added_mass>` tag based method[http://sdformat.org/spec?ver=1.11&elem=link#inertial_fluid_added_mass]To get rid of this warning we recommend setting`<disable_added_mass> to true.
> ```.
> ```

Error has been solved! Thanks for testing.

closed 2024-09-03 by arjo129