gazebosim/gazebo_test_cases #1592: gz-sim: boundingbox_camera.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/boundingbox_camera.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-08-29:
Interested, have experience with Object detector and visualizing them in 2D

akky20, 2024-08-31:
## No issue found
[image]

Mechazo11, 2024-09-02:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full Desktop 
    ```
* **Summary**
    * ```boundingbox_camera.sdf``` works with a number of warning messages shown.

https://github.com/user-attachments/assets/127f8e82-0800-4c7e-aa1d-7382fa6d4328

* **Issues**
    * Warning messages
    ```terminal
    libEGL warning: egl: failed to create dri2 screen
    libEGL warning: egl: failed to create dri2 screen
    libEGL warning: egl: failed to create dri2 screen
    Warning [Utils.cc:132] [/sdf/plugin[@name="gz::sim::systems::Label"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].
    Warning [Utils.cc:132] [/sdf/plugin[@name="gz::sim::systems::Label"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf].
    libEGL warning: egl: failed to create dri2 screen
    libEGL warning: egl: failed to create dri2 screen
    libEGL warning: egl: failed to create dri2 screen
    Warning [Utils.cc:132] [/sdf/plugin[@name="gz::sim::systems::Label"]:<data-string>:L1]: XML Element[plugin], child of element[sdf], not defined in SDF. Copying[plugin] as children of [sdf]
    ```
    
* **Fixes**
    * N/A

* **Comments/Suggestions**
    * N/A

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde