gazebosim/gazebo_test_cases #1593: gz-sim: breadcrumbs.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/breadcrumbs.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-09-02:
## For me no error is there , Works fine 

[Screencast from 02-09-24 11:53:08 PM IST.webm](https://github.com/user-attachments/assets/a2749458-c7d7-4146-8697-c2a9f07e2cf8)

closed 2024-09-03 by arjo129

iche033, 2024-09-03:
I can reproduce the issue with `[error] Failed to make breadcrumb 'B2_0' static.`, will look into it

iche033, 2024-09-04:
https://github.com/gazebosim/gz-sim/pull/2593