gazebosim/gazebo_test_cases #1594: gz-sim: buoyancy.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/buoyancy.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-31:
Interested

akky20, 2024-08-31:
## Here is Demo , no issues
[Screencast from 01-09-24 02:02:07 AM IST.webm](https://github.com/user-attachments/assets/77812daf-c488-48d2-8ed9-3164a07fe991)

khshmt, 2024-08-31:
## Ubuntu 22.04 (jammy) ##
[Screencast from 08-31-2024 11:52:57 PM.webm](https://github.com/user-attachments/assets/d6db5929-591a-477c-bbd9-5a5bcf2a087b)

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde