gazebosim/gazebo_test_cases #1595: gz-sim: buoyancy_engine.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/buoyancy_engine.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-31:
Interested

khshmt, 2024-08-31:
## Ubuntu 22.04 (jammy)
[Screencast from 08-31-2024 06:28:01 PM.webm](https://github.com/user-attachments/assets/c90d55d4-bf96-4825-b93b-29aa16e55804)
[Screencast from 08-31-2024 06:29:50 PM.webm](https://github.com/user-attachments/assets/66e0e975-3b40-4f86-aee7-31118b5dd244)
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde