gazebosim/gazebo_test_cases #1596: gz-sim: camera_lens_flare.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/camera_lens_flare.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-30:
Interested

khshmt, 2024-08-30:
## Ubuntu 22.04 (jammy) ##
[image]
[image]

but this is the terminal logging
[image]

iche033, 2024-08-30:
created https://github.com/gazebosim/gz-sim/pull/2559 to fix the issue.

khshmt, 2024-08-31:
> created [gazebosim/gz-sim#2559](https://github.com/gazebosim/gz-sim/pull/2559) to fix the issue.

It Works fine now
[image]

ahcorde, 2024-09-03:
Thank you for testing! and for the fix @iche033 !

closed 2024-09-03 by ahcorde