gazebosim/gazebo_test_cases #1598: gz-sim: camera_video_record_dbl_pendulum.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/camera_video_record_dbl_pendulum.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-31:
Interested

khshmt, 2024-08-31:
## Ubuntu 22.04 (jammy) ##

This the saved video from running the sdf.
https://github.com/user-attachments/assets/739a43ba-0575-41a1-a705-4941de327d29

khshmt, 2024-08-31:
[Screencast from 08-31-2024 03:59:53 PM.webm](https://github.com/user-attachments/assets/2d9e58a4-7ae6-412e-a815-a3f596aa85f2)

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde