gazebosim/gazebo_test_cases #1600: gz-sim: contact_sensor.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/contact_sensor.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-31:
Interested

akky20, 2024-08-31:
### Hello I have done this please check 

## Before Simulation
[image]

## After Simulation
[image]

When running the topic
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde