gazebosim/gazebo_test_cases #1604: gz-sim: dem_monterey_bay.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/dem_monterey_bay.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-31:
Interested

khshmt, 2024-08-31:
## Ubuntu 22.04 (jammy) ##
[image]
[image]

david-dorf, 2024-08-31:
Encountered the same issue with Ogre

azeey, 2024-09-16:
Fix in https://github.com/gazebosim/gz-gui/pull/642

closed 2024-09-16 by azeey