gazebosim/gazebo_test_cases #1605: gz-sim: dem_moon.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/dem_moon.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

david-dorf, 2024-08-31:
[Screencast from 2024-08-31 14-58-08.webm](https://github.com/user-attachments/assets/07977f89-99e3-4a6f-8f93-b8eb100c5666)
Texture/lighting error on launch. Otherwise looks okay

Mechazo11, 2024-09-01:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco full desktop installation
    ```
* **Summary**
    * ```dem_moon.sdf``` world opens successfully with some texture/shadow errors but it can **only be done** so from GUI. Direct command line launch does not work.

[image]

* **Issues**
    * Texture and lighting error as shown below in the screenshot
    ```bash
    (2024-09-01 15:30:14.597) [error] [GUI] Setting shadow texture size not supported by: ogre
    (2024-09-01 15:30:14.597) [error] [GUI] Shadow texture size not supported by: ogre
    (2024-09-01 15:30:14.597) [error] [GUI] Unable to set directional light shadow <texture_size> to '2048'. Using default texture size of 0
    ```
    
* **Fixes**
    * [BUG] ```gz sim -s dem_moon.sdf``` does not work, the simulation only works if I start from ```gz sim``` and then manually select the world to open.
    * ```gdb ruby``` and then  ```r $HOME/workspace/install/bin/gz sim -s dem_moon.sdf``` does not work either, the world does not launch

* **Comments/Suggestions**
    * N/A

Mechazo11, 2024-09-04:
@iche033, @azeey-test I think this one can be closed.

closed 2024-09-04 by iche033