gazebosim/gazebo_test_cases #1608: gz-sim: detachable_joint.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/detachable_joint.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

akky20, 2024-08-31:
## I have tested and has no issues 
[Screencast from 01-09-24 01:44:04 AM IST.webm](https://github.com/user-attachments/assets/d346a095-cebc-4bdb-ab45-b4849975312a)
also for the state it give the message if attached or not , unfortunatly I'm not able to record it but you can see it in terminal

avanmalleghem, 2024-09-01:
@akky20 please avoid doing it before saying you are interested in... So that we don't do the job twice...

arjo129, 2024-09-03:
Thanks for testing!

closed 2024-09-03 by arjo129