gazebosim/gazebo_test_cases #1609: gz-sim: diff_drive.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/diff_drive.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

EnzoGhisoni, 2024-08-30:
I am interested to test this

david-dorf, 2024-08-31:
[Screencast from 2024-08-31 14-55-28.webm](https://github.com/user-attachments/assets/ebc0802a-98da-4cd1-aa7e-2bd5baedd4ba)
LGTM

EnzoGhisoni, 2024-08-31:
Step 1 - Instructions work

The sdf file is working as expected, to test it:
- create diff_drive.sdf and copy/paste the code
- run it with the command: gz sim diff_drive.sdf
- run the terminal command to publish on cmd_vel topics (result in video)
- run the echo commands to check the odom publication (image with terminal log of the odometry

Step 2 - Images (if there are any) match the result
Video of diff_drive robots running:
[DiffDrive.webm](https://github.com/user-attachments/assets/edd3caf7-59f3-49ea-95ca-1ed97392de1a)

Odometry terminal log:
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde