gazebosim/gazebo_test_cases #1614: gz-sim: empty_gui.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/empty_gui.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

avanmalleghem, 2024-08-31:
Issue when running the first command. The world is not empty

[image]

arjo129, 2024-09-03:
Yes please update the SDF file!

avanmalleghem, 2024-09-03:
After more investigation, I think this SDF makes no-sense with this version. In fact, everything is set up correctly because of the `.gz/sim/9/gui.config`. I guess it has changed since the latest version (now it adds SelectEntities, InteractiveControl, CameraTracking, VisualizationCapabilities and GzSceneManager plugins). So all the tutorial in this file doesn't make sense anymore.

I think the best option is to remove it completely. If ok for you, I can open a PR.

iche033, 2024-09-03:
oh good catch, that's a behavior change in newer versions. This should prevent all other plugins from being loaded:

```diff
diff --git a/examples/worlds/empty_gui.sdf b/examples/worlds/empty_gui.sdf
index 0975a4f14..54df8c2e3 100644
--- a/examples/worlds/empty_gui.sdf
+++ b/examples/worlds/empty_gui.sdf
@@ -37,6 +37,10 @@ This example helps illustrate the interaction of the MinimalScene with other GUI
 
   <world name="empty_gui">
 
+    <gz:policies>
+      <include_gui_default_plugins>false</include_gui_default_plugins>
+    </gz:policies>
+
     <gui fullscreen="0">
       <plugin filename="MinimalScene" name="3D View">
         <gz-gui>
```

Can you see if that works for you? If so, we can keep this demo world

avanmalleghem, 2024-09-04:
Indeed, it fix the issue. I created a PR : https://github.com/gazebosim/gz-sim/pull/2591

closed 2024-09-04 by iche033