gazebosim/gazebo_test_cases #1618: gz-sim: fuel.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/fuel.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-30:
Interested

khshmt, 2024-08-30:
## Ubuntu 22.04 (jammy) ##
[image]
[image]

khshmt, 2024-08-30:
[Screencast from 08-30-2024 09:36:35 PM.webm](https://github.com/user-attachments/assets/98855d8c-8d16-49eb-bf6f-6419da7de2bb)

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde