gazebosim/gazebo_test_cases #1621: gz-sim: gpu_lidar_sensor.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/gpu_lidar_sensor.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-08-29:
Interested, have RTX 3080 hence should not be resource constrained.

akky20, 2024-08-31:
No issues 
[Screencast from 01-09-24 01:52:56 AM IST.webm](https://github.com/user-attachments/assets/be640645-c60f-48d6-be2d-74db5254ef72)

Mechazo11, 2024-09-01:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full Desktop 
    ```
* **Summary**
    * ```gpu_lidar_sensor.sdf``` works but console output shows very large numbers intermittently. See video

https://github.com/user-attachments/assets/67635e95-fb41-4f96-9d31-2e714fa3c1b0

* **Issues**
    * N/A
    
* **Fixes**
    * N/A

* **Comments/Suggestions**
    * N/A

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde