gazebosim/gazebo_test_cases #1624: gz-sim: ground_spacecraft_testbed.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/ground_spacecraft_testbed.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

khshmt, 2024-08-30:
Interested

khshmt, 2024-08-30:
## Ubuntu 22.04 (jammy) ##
[image]

khshmt, 2024-08-31:
Interested

khshmt, 2024-08-31:
## Ubuntu 22.04 (jammy) ##
the space craft does not move.
[image]

Mechazo11, 2024-09-01:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full Desktop 
    ```
* **Summary**
    * ```ground_spacecraft_testbed.sdf``` opens correctly but the tutorial had wrong thrust command. Changed command to the correct one. Opened a PR request. The robot moves correctly as shown below

https://github.com/user-attachments/assets/7c37e46d-62a3-4c9f-bb58-1ba6502d93af

* **Issues**
    * Ground robot does not move with the given command.
    
* **Fixes**
    * ```gz topic -p 'normalized:[1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]' -t /kth_freeflyer/command/duty_cycle --msgtype gz.msgs.Actuators```
    
    * See PR request [#2565](https://github.com/gazebosim/gz-sim/pull/2565)

* **Comments/Suggestions**
    * N/A

arjo129, 2024-09-02:
Thanks for fixing the docs @Mechazo11. Closing as we have merged the fix.

closed 2024-09-02 by arjo129