gazebosim/gazebo_test_cases #1627: gz-sim: joint_controller.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/joint_controller.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

akky20, 2024-08-31:
## NO Issues , I have attached some images for reference 
[image]
[image]
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde