gazebosim/gazebo_test_cases #1628: gz-sim: joint_position_controller.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/joint_position_controller.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

akky20, 2024-08-31:
## Spawing without simulation
[image]

## With the Topics
[image]
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde