gazebosim/gazebo_test_cases #1629: gz-sim: joint_trajectory_controller.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/joint_trajectory_controller.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Combuster54, 2024-08-30:
Interested

Combuster54, 2024-08-31:
Running, I'm facing some warnings
```bash
gz  sim joint_trajectory_controller.sdf
```

### Warnings 

```bash
Warning [Utils.cc:132] [/sdf/world[@name="default"]/model[@name="RR_velocity_control"]/joint[@name="RR_velocity_control_joint1"]/dynamics:/usr/share/gz/gz-sim9/worlds/joint_trajectory_controller.sdf:L483]: XML Element[dynamics], child of element[joint], not defined in SDF. Copying[dynamics] as children of [joint].
Warning [Utils.cc:132] [/sdf/world[@name="default"]/model[@name="RR_velocity_control"]/joint[@name="RR_velocity_control_joint2"]/dynamics:/usr/share/gz/gz-sim9/worlds/joint_trajectory_controller.sdf:L494]: XML Element[dynamics], child of element[joint], not defined in SDF. Copying[dynamics] as children of [joint].
```

`<dynamics>` tags should be inside `<axis>` 

URL: http://sdformat.org/spec?ver=1.6&elem=joint

```xml
    line 475 
      <!-- Joints -->
      <joint name="RR_velocity_control_joint1" type="revolute">
        <pose relative_to="RR_velocity_control_link0">0 0 0 0 0 0</pose>
        <parent>RR_velocity_control_link0</parent>
        <child>RR_velocity_control_link1</child>
        <axis>
          <xyz>1 0 0</xyz>
        </axis>
        <dynamics>
          <damping>0.02</damping>
        </dynamics>
      </joint>
      <joint name="RR_velocity_control_joint2" type="revolute">
        <pose relative_to="RR_velocity_control_link1">0 0 0.1 0 0 0</pose>
        <parent>RR_velocity_control_link1</parent>
        <child>RR_velocity_control_link2</child>
        <axis>
          <xyz>1 0 0</xyz>
        </axis>
        <dynamics>
          <damping>0.01</damping>
        </dynamics>
      </joint>
```

### Solution

Reposition the dynamics tags of the joints RR_velocity_control_joint1 and RR_velocity_control_joint2 inside the axis tag.

```xml
    line 475 
      <!-- Joints -->
      <joint name="RR_velocity_control_joint1" type="revolute">
        <pose relative_to="RR_velocity_control_link0">0 0 0 0 0 0</pose>
        <parent>RR_velocity_control_link0</parent>
        <child>RR_velocity_control_link1</child>
        <axis>
          <xyz>1 0 0</xyz>
          <dynamics>
            <damping>0.01</damping>
          </dynamics>
        </axis>
      </joint>
      <joint name="RR_velocity_control_joint2" type="revolute">
        <pose relative_to="RR_velocity_control_link1">0 0 0.1 0 0 0</pose>
        <parent>RR_velocity_control_link1</parent>
        <child>RR_velocity_control_link2</child>
        <axis>
          <xyz>1 0 0</xyz>
          <dynamics>
            <damping>0.01</damping>
          </dynamics>
        </axis>
      </joint>
```

## Testing joint trajectory commands

**Position control (Red RR robot)**

[image]

**Velocity control (Green RR robot)**

[image]

**Effort control (Blue RR robot)**

[image]

ahcorde, 2024-09-02:
PR https://github.com/gazebosim/gz-sim/pull/2580

ahcorde, 2024-09-03:
thank you for testing! PR is merged

closed 2024-09-03 by ahcorde