gazebosim/gazebo_test_cases #1633: gz-sim: lift_drag.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/lift_drag.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-08-31:
No Issues, Here is the images for the reference 

[image]
[image]
[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde