gazebosim/gazebo_test_cases #1640: gz-sim: log_record_dbl_pendulum.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/log_record_dbl_pendulum.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

avanmalleghem, 2024-08-31:
It seems like `gz sim --playback ~/.gz/sim/log/<timestamp>` is not working. Maybe because I run it into your docker setup ? Otherwise, the record seems ok
[image]

For example, I launched `gz sim -v 4 --playback ~/.gz/sim/log/2024-08-31T14\:06\:29.302105487`and it doesn't work and don't show any interesting error logs.

arjo129, 2024-09-03:
Yes. I can confirm the playback is not working.

avanmalleghem, 2024-09-03:
@arjo129 is it a docker related issue or even on an ubuntu host ? How can we progress on this one ?

arjo129, 2024-09-03:
It is an issue on the host as well. Im currently investigating.

arjo129, 2024-09-04:
After some insane Spelunking of the code base there is a bug in the way the GUI and the server synchronize the world name. The GUI client queries what it thinks is the world name. The server sends a world name that does not match what the world name in the log file is.

Evidence of this issue can be seen in this screenshot:

arjo129, 2024-09-04:
PR opened: https://github.com/gazebosim/gz-sim/pull/2590

avanmalleghem, 2024-09-06:
Isn't the issue related to the way gui config is managed ? I see in https://gazebosim.org/api/sim/9/gui_config.html : "For log-playback, the default file is $HOME/.gz/sim/<#>/playback_gui.config".

azeey, 2024-09-06:
hmm, we should check if this is related to https://github.com/gazebosim/gz-sim/pull/2497

arjo129, 2024-09-06:
@azeey I dont think this has to do with config files. Ive got a full fiz for this already in gazebosim/gz-sim#2590. Its a regression becaise of refactors to `SimulationRunner`

closed 2024-09-12 by azeey