gazebosim/gazebo_test_cases #1643: gz-sim: log_record_shapes.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/log_record_shapes.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

avanmalleghem, 2024-08-31:
Interested in !

avanmalleghem, 2024-08-31:
Waiting for https://github.com/gazebosim/gazebo_test_cases/issues/1640

utkuesen, 2024-09-07:
I tried it with these commands.
gz sim log_record_shapes.sdf
gz sim -v 4 log_record_shapes.sdf
log files are created but I couldn't replay with below command.
gz sim --playback ~/.gz/sim/log/2024-09-08T04:54:53.338219518/state.tlog

Gazebo started but there was no object in the scene.
There was a error shown in terminal.
It said "Please specify a zip file".
When I pressed the play button, nothing is happened except counting simulation time.

Terminal outputs are below:
gz sim --playback ~/.gz/sim/log/2024-09-08T04:54:53.338219518/state.tlog
(2024-09-08 04:56:26.842) [error] Please specify a zip file.
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen

Screen:
[image]

avanmalleghem, 2024-09-09:
@utkuesen please, follow the process available here https://community.gazebosim.org/t/gazebo-ionic-test-and-tutorial-party-instructions/2989. This issue is already assigned. And is related to another issue because of a related bug we need to solve before being able to continue on this one.

avanmalleghem, 2024-09-13:
It works as expected 
[image]

closed 2024-09-13 by azeey