gazebosim/gazebo_test_cases #1646: gz-sim: lrauv_control_demo.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/lrauv_control_demo.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alesof, 2024-08-30:
Interested ✋🏼

alesof, 2024-08-30:
## Configuration
Running on Ubuntu 22.04 using the suggested docker setup. Gazebo version is 9.0.0~pre1.

## Process
Step1: The simulation works but an error message is displayed:
```
[error] The use of added mass through this plugin is deprecated and willbe removed in Gazebo J* as this formulation has instabilities. We recommend using the SDF `<fluid_added_mass>` tag based method[http://sdformat.org/spec?ver=1.11&elem=link#inertial_fluid_added_mass]To get rid of this warning we recommend setting`<disable_added_mass> to true.
```

To test the simulation:
```
gz topic -t /model/tethys/joint/horizontal_fins_joint/0/cmd_pos -m gz.msgs.Double -p 'data: -0.5'
gz topic -t /model/tethys/joint/vertical_fins_joint/0/cmd_pos -m gz.msgs.Double -p 'data: -0.5'
gz topic -t /model/tethys/joint/propeller_joint/cmd_thrust -m gz.msgs.Double -p 'data: -10'
```

[Opened a PR](https://github.com/gazebosim/gz-sim/pull/2558) that includes the recommend setting, thus removing the warning message.

Image displaying the running simulation:
[image]

arjo129, 2024-09-03:
Fixed warning by: https://github.com/gazebosim/gz-sim/pull/2579 and ticketed model re-upload here: https://github.com/gazebosim/gz-sim/issues/2575

Thanks for the help @alesof

closed 2024-09-03 by arjo129