gazebosim/gazebo_test_cases #1647: gz-sim: mecanum_drive.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/mecanum_drive.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

Mechazo11, 2024-08-29:
Interested.

Mechazo11, 2024-09-01:
* **Test enviornment**
    ```text
    CPU: Ryzen 5 5600x, Ram: 32GB @ 3200 Mhz, GPU: RTX 3080
    Clean copy of Ubuntu 24.04 Humble
    ROS 2 Jazzy Jalisco Full Desktop 
    ```
* **Summary**
    * ```mecanum_drive.sdf``` world opens successfully with no issue. See video.

https://github.com/user-attachments/assets/eb377abf-ecdd-46d4-98c4-998d1293fdd5

* **Issues**
    * N/A
    
* **Fixes**
    * N/A

* **Comments/Suggestions**
    * N/A

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde