gazebosim/gazebo_test_cases #1650: gz-sim: model_photo_shoot.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/model_photo_shoot.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

alesof, 2024-08-30:
Interested ✋🏼

alesof, 2024-08-30:
## Configuration
Running on Ubuntu 22.04 using the suggested docker setup. Gazebo version is 9.0.0~pre1.

## Process
Step1: The simulation works. Several debug outputs are printed to console, don't know if it is the desired behaviour. 

To test the simulation:
```
gz sim  -s -r -v 4 \-\-iterations 50 model_photo_shoot.sdf
```

Produced output matches the description and attached.

[image]
[image]
[image]
[image]
[image]

closed 2024-08-30 by iche033