gazebosim/gazebo_test_cases #1652: gz-sim: multicopter_velocity_control.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/multicopter_velocity_control.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-08-31:
## When we spawn the models
[image]

## Start the simulation
[image]

## when we give command to quadcopter to straight up and then hover
[image]
[image]

## odometry and pose result 
[image]
[image]

# Now for hexacopter
[image]
[image]

[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde