gazebosim/gazebo_test_cases #1655: gz-sim: optical_tactile_sensor_plugin.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/optical_tactile_sensor_plugin.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-08-30:
[image]

Spawing perfectly, but yes whenever I run simulation the gui stops with these console logs 
[image]

iche033, 2024-08-31:
created https://github.com/gazebosim/gz-sim/pull/2561 to fix the crash.

closed 2024-09-03 by iche033