gazebosim/gazebo_test_cases #1658: gz-sim: perfect_comms.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/perfect_comms.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

akky20, 2024-08-30:
I'm interest

akky20, 2024-08-30:
[image]

Works fine

azeey, 2024-08-30:
The instructions in the .SDF file say
  Gazebo perfect-comms plugin demo.

 
>  Compile the comms publisher:
>     cd gz-sim/examples/standalone/comms
>     mkdir build
>     cd build
>     cmake ..
>     make
> 
>   Try launching a comms subscriber:
>     gz topic -e -t addr2/rx
> 
>   Try launching a comms publisher:
>     ./publisher addr2

Can you confirm if you followed those steps?

akky20, 2024-08-30:
[image]

yes here it is , but for subscribing I use `gz topic -e -t /broker/msgs` as with `gz topic -e -t addr2/rx
` its not subscribing anything

closed 2024-09-04 by iche033