gazebosim/gazebo_test_cases #1670: gz-sim: rolling_shapes.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/rolling_shapes.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

david-dorf, 2024-08-31:
They be rolling, even after the visual plane ends... Maybe a problem with the Plane geometry? The SDF of the world looked fine
[Screencast from 2024-08-31 14-46-59.webm](https://github.com/user-attachments/assets/81de046b-7050-4bba-8cef-a346643cc8ba)

ahcorde, 2024-09-02:
The plane is not infinite. Thank you for testing!

closed 2024-09-02 by ahcorde