gazebosim/gazebo_test_cases #1672: gz-sim: sensors.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/sensors.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

trushant05, 2024-08-30:
## Gazebo sensor demo for sensors that do not require rendering

## Altimeter: 

```gz topic -e -t /altimeter```

[image]

##  Imu: 

```gz topic -e -t /imu```

[image]

##  Magnetometer: 

```gz topic -e -t /magnetometer```

[image]

##  Air pressure: 

```gz topic -e -t /air_pressure```

[image]

ahcorde, 2024-09-02:
Thank you for testing!

closed 2024-09-02 by ahcorde