gazebosim/gazebo_test_cases #1676: gz-sim: shapes_bitmask.sdf

azeey-test, 2024-08-29:
## Configuration

* os: Any

## Links

* [Source](https://github.com/gazebosim/gz-sim/tree/gz-sim9/examples/worlds/shapes_bitmask.sdf)

## Process

    
            
### Step 1 - Instructions work
            
        

    
    

    

    
    
            
### Step 2 - Images (if there are any) match the result

bperseghetti, 2024-08-29:
I'm available to test.

david-dorf, 2024-08-31:
[Screencast from 2024-08-31 14-36-30.webm](https://github.com/user-attachments/assets/2a61280e-f8db-4b76-9398-d584d2b454f8)
Generally worked (Red and Green did not collide but both did with Yellow), however moving the box underneath Green didn't result in Green falling due to gravity. This probably isn't the plugin itself, but a problem with the sim physics itself. No meaningful errors or warnings occurred.

azeey, 2024-09-16:
@david-dorf I was unable to reproduce what I saw in the video. If you're able to reproduce it, could you please open an issue on gz-sim? I'll go ahead and close this issue since the transform tool problem is a separate issue.

closed 2024-09-16 by azeey

iche033, 2024-09-16:
I also couldn't reproduce. Looks like somehow sim object poses ended up in a bad state.

david-dorf, 2024-09-16:
I’ll take another look when I get home, I agree it’s probably the transform
tool.